ARMmbed/mbed-ls #359: mbedls 1.4.6 hangs on macOS 10.13.5 Python3.6

cederom, 2018-06-24:
Hello world :-) I have noticed a problem with `mbedls` on macOS and Python 3.6.5 virtualenv, it hangs with no results.. this also affects other tools that depend on `mbed-ls` :-)

```
(venv36) mbedls --version
1.4.6

(venv36) mbedls -d
DEBUG:mbedls.main:mbed-ls ver. 1.4.6
DEBUG:mbedls.main:host: ('posix', 'Darwin', '17.6.0', 'Darwin Kernel Version 17.6.0: Tue May  8 15:22:16 PDT 2018; root:xnu-4570.61.1~1/RELEASE_X86_64', 'darwin')

<<nothing shows here>>

(venv36) python --version
Python 3.6.5
```

Seems to work well in Python 2.7 virtualenv:
```
(venv27) $ mbedls --version
1.3.7
(venv27) $ mbedls
+---------------+----------------------+------------------+-----------------------+--------------------------------------------------+-----------------+
| platform_name | platform_name_unique | mount_point      | serial_port           | target_id                                        | daplink_version |
+---------------+----------------------+------------------+-----------------------+--------------------------------------------------+-----------------+
| unknown       | None[0]              | /Volumes/DAPLINK | /dev/tty.usbmodem1412 | 110100004420312033504139313031313233303097969903 | 0244            |
+---------------+----------------------+------------------+-----------------------+--------------------------------------------------+-----------------+

(venv27) $ python --version
Python 2.7.14
```

```
$ mbedls --version
WARNING:mbedls.platform_database:Error loading database /Users/cederom/Library/Application Support/mbedls/platforms.json: Platform Database is out of date; Recreating
1.4.6
(venv27) $ mbedls
+---------------+----------------------+------------------+-----------------------+--------------------------------------------------+-----------------+
| platform_name | platform_name_unique | mount_point      | serial_port           | target_id                                        | daplink_version |
+---------------+----------------------+------------------+-----------------------+--------------------------------------------------+-----------------+
| NRF52_DK      | NRF52_DK[0]          | /Volumes/DAPLINK | /dev/tty.usbmodem1412 | 110100004420312033504139313031313233303097969903 | 0244            |
+---------------+----------------------+------------------+-----------------------+--------------------------------------------------+-----------------+

$ python --version
Python 2.7.15
```

bridadan, 2018-06-25:
Huh, weird! Thanks for raising the issue. @theotherjimmy you have a Mac to test on right?

cederom, 2018-06-25:
Always welcome @bridadan let me know if you need some more testing :-) I just heard about Python3 support so I wanted to try that out.. while trying `mbed sterm` did not get anything so I thought that would depend on `mbed-ls` and got into this issue.. in the meantime I am using `minicom` for serial console and that works fine.. so maybe a detection phase issue..? :-)

theotherjimmy, 2018-06-25:
Maybe. I'll take a look.

theotherjimmy, 2018-06-25:
Confirmed. It's getting stuck in an iosomething.wait()

jupe, 2018-08-13:
this happens also with latest release:
```
$ mbedls -d
DEBUG:mbedls.main:mbed-ls ver. 1.5.1
DEBUG:mbedls.main:host: ('posix', 'Darwin', '17.7.0', 'Darwin Kernel Version 17.7.0: Thu Jun 21 22:53:14 PDT 2018; root:xnu-4570.71.2~1/RELEASE_X86_64', 'darwin')
^CTraceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.6/bin/mbedls", line 11, in <module>
    load_entry_point('mbed-ls==1.5.1', 'console_scripts', 'mbedls')()
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/mbed_lstools/main.py", line 262, in mbedls_main
    ret_code = args.command(mbeds, args)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/mbed_lstools/main.py", line 111, in print_table
    return print_mbeds(mbeds, args, False)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/mbed_lstools/main.py", line 98, in print_mbeds
    devices = mbeds.list_mbeds(unique_names=True, read_details_txt=True)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/mbed_lstools/lstools_base.py", line 136, in list_mbeds
    candidates = list(self.find_candidates())
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/mbed_lstools/darwin.py", line 106, in find_candidates
    volumes = self._volumes()
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/mbed_lstools/darwin.py", line 159, in _volumes
    ioreg_usb.wait()
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/subprocess.py", line 1436, in wait
    (pid, sts) = self._try_wait(0)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/subprocess.py", line 1383, in _try_wait
    (pid, sts) = os.waitpid(self.pid, wait_flags)
KeyboardInterrupt
```

Any chance to get fix for this ?

theotherjimmy, 2018-10-30:
I think this ([from here](https://docs.python.org/3/library/subprocess.html?highlight=subprocess#subprocess.Popen.wait)) is related:

>Note
>
>This will deadlock when using stdout=PIPE or stderr=PIPE and the child process generates enough output to a pipe such that it blocks waiting for the OS pipe buffer to accept more data. Use Popen.communicate() when using pipes to avoid that.

PR #399 (2018-11-05): Avoid deadlock on reading PList process output
On Mac (Darwin), we read much of the USB and MSD info from XML Plists.
These plists are parsed by python's `plistlib`. The prior implementation
would assume that the process generating the plist would complete by the
time the XML parser reached the end of it's current stream. As it turns
out, python3's XML parser violates this assumption. This change forces
the plist-generating process to complete before parsing starts. This may
be a tad slower, but it makes sure that we don't enter a state where we
have finished processing a plist, and the plist- [... cut]
closed 2018-11-05 by theotherjimmy