ARMmbed/mbed-ls #360: Can't detect target with JLink 8.0

OmerRosler, 2018-06-28:
We are using the newest JLink commander software and fail to detect our connected board.

The device is JLink plus:
[image]
We use this connection for flashing during debugging and directly from the command line, but fail the detection in mbed-ls.

`mbed-ls --debug` output:

DEBUG:mbedls.main:mbed-ls ver. 1.4.5
DEBUG:mbedls.main:host: ('nt', 'Windows', '7', '6.1.7601', 'win32')
DEBUG:mbedls.lstools_win7:_iter_vals <PyHKEY object at 0x026C87D70>
DEBUG:mbedls.lstools_win7:Fetching mounted devices from disk service registry entry
DEBUG:mbedls.lstools_win7:_iter_vals <PyHKEY object at 0x026C8D70>
DEBUG:mbedls.lstools_win7:Fetching usb storage devices from USBSTOR service registry entry
DEBUG:mbedls.lstools_win7:No USBSTOR service found, no device can be detected
DEBUG:mbedls.lstools_win7:target_id_usb_id -> mount_point mapping: {}
DEBUG:mbedls.lstools_base:Candidates for display []
DEBUG:mbedls.main:Return code: 0

bridadan, 2018-06-28:
Hi @OmerRosler. Mbed LS currently doesn't support external debuggers "out of the box". The JLink detection functionality only works for development boards with the specific JLink firmware flashed to the board's interface chip.

bridadan, 2019-01-07:
Closing this for now since this would require a significant rewrite of mbed-ls to accomplish this.

closed 2019-01-07 by bridadan