ARMmbed/mbed-ls #363: mbedls-1.4.6 assigns same target id for separate devices

paviitan, 2018-07-06:
### Description

I am having trouble with telling two development boards apart from each other. When I plug in just one, mbed-ls gives me:
```
NUCLEO_F411RE  NUCLEO_F411RE[0]  /media/mbed  /dev/ttyACM1  07400221014C650F3E08F372   0221
```

 After plugging in second board, both mbed-ls changes both target ids to match each other. 
```
NUCLEO_F411RE  NUCLEO_F411RE[0]  /media/mbed  /dev/ttyACM1  07400221014465673E00F31A  0221
NUCLEO_F411RE  NUCLEO_F411RE[1]  /media/mbed  /dev/ttyACM0  07400221014465673E00F31A  0221
```

#### mbed-ls version
```
1.4.6
```
#### OS
```
Distributor ID: Ubuntu
Description:    Ubuntu 18.04 LTS
Release:        18.04
Codename:       bionic
```
#### Device
```
NUCLEO_F411RE
```

### Debug log
```
mbedls -d
DEBUG:mbedls.main:mbed-ls ver. 1.4.6
DEBUG:mbedls.main:host: ('posix', 'Linux', '4.15.0-24-generic', '#26-Ubuntu SMP Wed Jun 13 08:44:47 UTC 2018', 'linux2')
DEBUG:mbedls.lstools_linux:Converting device list ['/dev/disk/by-id/usb-MBED_microcontroller_066FFF495648807567012226-0:0', '/dev/disk/by-id/usb-MBED_microcontroller_0668FF495648807567056042-0:0', '/dev/disk/by-id/wwn-0x5001480000000000', '/dev/disk/by-id/ata-hp_HLDS_DVDRW_GUD1N_B7LRVP9534243', '/dev/disk/by-id/wwn-0x5ace42e020269540-part1', '/dev/disk/by-id/wwn-0x5ace42e020269540-part2', '/dev/disk/by-id/ata-SKhynix_SC300_HFS256G32MND-3210A_ES7BN5180103ABF2W-part1', '/dev/disk/by-id/ata-SKhynix_SC300_HFS256G32MND-3210A_ES7BN5180103ABF2W-part2', '/dev/disk/by-id/wwn-0x5ace42e020269540', '/dev/disk/by-id/ata-SKhynix_SC300_HFS256G32MND-3210A_ES7BN5180103ABF2W', '/dev/disk/by-id/usb-Generic-_USB3.0_CRW_201408282030-0:0']
DEBUG:mbedls.lstools_linux:Found disk devices by id {'0668FF495648807567056042': '/dev/sdm', '066FFF495648807567012226': '/dev/sdn'}
DEBUG:mbedls.lstools_linux:Converting device list ['/dev/serial/by-id/usb-STMicroelectronics_STM32_STLink_066FFF495648807567012226-if02', '/dev/serial/by-id/usb-STMicroelectronics_STM32_STLink_0668FF495648807567056042-if02']
DEBUG:mbedls.lstools_linux:Found serial devices by id {'0668FF495648807567056042': '/dev/ttyACM1', '066FFF495648807567012226': '/dev/ttyACM0'}
DEBUG:mbedls.lstools_linux:Mount mapping {u'/dev/sdn': u'/media/mbed', u'/dev/sdm': u'/media/mbed', u'/dev/sda1': u'/boot/efi'}
DEBUG:mbedls.lstools_base:Candidates for display [{'mount_point': u'/media/mbed', 'serial_port': '/dev/ttyACM1', 'target_id_usb_id': '0668FF495648807567056042'}, {'mount_point': u'/media/mbed', 'serial_port': '/dev/ttyACM0', 'target_id_usb_id': '066FFF495648807567012226'}]
DEBUG:mbedls.lstools_base:Found target id 07400221014465673E00F31A in htm line <meta http-equiv="refresh" content="0; url=http://mbed.org/device/?code=07400221014465673E00F31A"/>

DEBUG:mbedls.lstools_base:Found htm target id, 07400221014465673E00F31A, for usb target id 0668FF495648807567056042
DEBUG:mbedls.lstools_base:Found target id 07400221014465673E00F31A in htm line <meta http-equiv="refresh" content="0; url=http://mbed.org/device/?code=07400221014465673E00F31A"/>

DEBUG:mbedls.lstools_base:Found htm target id, 07400221014465673E00F31A, for usb target id 066FFF495648807567012226
+---------------+----------------------+-------------+--------------+--------------------------+-----------------+
| platform_name | platform_name_unique | mount_point | serial_port  | target_id                | daplink_version |
+---------------+----------------------+-------------+--------------+--------------------------+-----------------+
| NUCLEO_F411RE | NUCLEO_F411RE[0]     | /media/mbed | /dev/ttyACM1 | 07400221014465673E00F31A | 0221            |
| NUCLEO_F411RE | NUCLEO_F411RE[1]     | /media/mbed | /dev/ttyACM0 | 07400221014465673E00F31A | 0221            |
+---------------+----------------------+-------------+--------------+--------------------------+-----------------+
DEBUG:mbedls.main:Return code: 0
```

bridadan, 2018-07-09:
Hi @paviitan, thanks for the detailed report!

In the debug log, I see that both of the devices are being detected as mounted to the same mount point:

```
DEBUG:mbedls.lstools_linux:Mount mapping {u'/dev/sdn': u'/media/mbed', u'/dev/sdm': u'/media/mbed', u'/dev/sda1': u'/boot/efi'}
DEBUG:mbedls.lstools_base:Candidates for display [{'mount_point': u'/media/mbed', 'serial_port': '/dev/ttyACM1', 'target_id_usb_id': '0668FF495648807567056042'}, {'mount_point': u'/media/mbed', 'serial_port': '/dev/ttyACM0', 'target_id_usb_id': '066FFF495648807567012226'}]
```

You can see that both devices `/dev/sdn` and `/dev/sdm` are mounted to `/media/mbed`. Can you confirm that this is correct on your machine? If this is true, then this will cause the issue you're seeing above. If you mount them on two separate mount points then the issue should be fixed.

**If they aren't actually mounted to the same mount point**, could you please post the output of the `mount` command here? Thanks!

bridadan, 2018-07-12:
Hi @paviitan, have you had any luck with your issue?

paviitan, 2018-07-13:
@bridadan, I will check this again and get back to you early next week.

paviitan, 2018-07-16:
@bridadan I was able to mount the devices to different mount points and that fixed the issue. Thank you!

closed 2018-07-16 by paviitan