ARMmbed/mbed-ls #366: two results has same serial_port

jupe, 2018-07-13:
For some reason I got two board results with same serial_port. Second board is probably in bad shape and should not be listed - at least with same serial port. Maybe results should be verified that each record has uniq mount_point, serial_port and target_id before giving to client.

Logs below:
```
DEBUG:mbedls.main:mbed-ls ver. 1.4.6
DEBUG:mbedls.main:host: ('posix', 'Darwin', '17.6.0', 'Darwin Kernel Version 17.6.0: Tue May  8 15:22:16 PDT 2018; root:xnu-4570.61.1~1/RELEASE_X86_64', 'darwin')
DEBUG:mbedls.lstools_darwin:finding in 
{ 'IORegistryEntryChildren': [ { 'IORegistryEntryName': 'AppleUSBHostLegacyClient'},
                               { 'IORegistryEntryName': 'AppleUSBiBridge',
                                 'idProduct': 34304,
                                 'idVendor': 1452},
                               { 'IORegistryEntryName': 'FaceTime HD Camera (Built-in, SN:CC2710351NSGJJM1D)',
                                 'idProduct': 34304,
                                 'idVendor': 1452},
                               { 'IORegistryEntryName': 'IOUSBHostInterface',
                                 'idProduct': 34304,
                                 'idVendor': 1452},
                               { 'IORegistryEntryChildren': [ { 'IORegistryEntryChildren': [ { 'IORegistryEntryChildren': [ { 'IORegistryEntryChildren': [ { 'IORegistryEntryName': 'IOHIDEventServiceUserClient'}],
                                                                                                                              'IORegistryEntryName': 'IOHIDEventDriver'}],
                                                                                               'IORegistryEntryName': 'IOHIDInterface'},
                                                                                             { 'IORegistryEntryName': 'IOHIDLibUserClient'}],
                                                                'IORegistryEntryName': 'IOUSBHostHIDDevice'}],
                                 'IORegistryEntryName': 'IOUSBHostInterface',
                                 'idProduct': 34304,
                                 'idVendor': 1452},
                               { 'IORegistryEntryName': 'IOUSBHostInterface',
                                 'idProduct': 34304,
                                 'idVendor': 1452},
                               { 'IORegistryEntryChildren': [ { 'IORegistryEntryName': 'AppleUSBNCMControl'}],
                                 'IORegistryEntryName': 'NCM Control',
                                 'idProduct': 34304,
                                 'idVendor': 1452},
                               { 'IORegistryEntryChildren': [ { 'IORegistryEntryChildren': [ { 'BSD Name': 'en5',
                                                                                               'IORegistryEntryChildren': [ { 'IORegistryEntryChildren': [ { 'IORegistryEntryName': 'IONetworkStackUserClient'}],
                                                                                                                              'IORegistryEntryName': 'IONetworkStack'}],
                                                                                               'IORegistryEntryName': 'en5'}],
                                                                'IORegistryEntryName': 'AppleUSBNCMData',
                                                                'idProduct': 34304,
                                                                'idVendor': 1452}],
                                 'IORegistryEntryName': 'NCM Data',
                                 'idProduct': 34304,
                                 'idVendor': 1452},
                               { 'IORegistryEntryChildren': [ { 'IORegistryEntryChildren': [ { 'IORegistryEntryChildren': [ { 'IORegistryEntryName': 'AppleUSBALSService'}],
                                                                                               'IORegistryEntryName': 'IOHIDInterface'},
                                                                                             { 'IORegistryEntryName': 'IOHIDLibUserClient'},
                                                                                             { 'IORegistryEntryName': 'IOHIDLibUserClient'}],
                                                                'IORegistryEntryName': 'IOUSBHostHIDDevice'}],
                                 'IORegistryEntryName': 'IOUSBHostInterface',
                                 'idProduct': 34304,
                                 'idVendor': 1452},
                               { 'IORegistryEntryChildren': [ { 'IORegistryEntryName': 'KernelRelayHost',
                                                                'idVendor': 1452}],
                                 'IORegistryEntryName': 'Apple USB SEP Interface',
                                 'idProduct': 34304,
                                 'idVendor': 1452}],
  'IORegistryEntryName': 'iBridge',
  'idProduct': 34304,
  'idVendor': 1452}
DEBUG:mbedls.lstools_darwin:finding in 
{ 'IORegistryEntryChildren': [ { 'IORegistryEntryName': 'AppleUSBHostLegacyClient'},
                               { 'IORegistryEntryChildren': [ { 'IORegistryEntryChildren': [ { 'IORegistryEntryChildren': [ { 'IORegistryEntryName': 'AppleUSBHostLegacyClient'},
                                                                                                                            { 'IORegistryEntryName': 'AppleUSBHostCompositeDevice'},
                                                                                                                            { 'IORegistryEntryChildren': [ { 'IORegistryEntryChildren': [ { 'IORegistryEntryChildren': [ { 'IORegistryEntryName': 'AppleUSBAudioStream',
                                                                                                                                                                                                                           'idProduct': 2616,
                                                                                                                                                                                                                           'idVendor': 1133},
                                                                                                                                                                                                                         { 'IORegistryEntryName': 'AppleUSBAudioStream',
                                                                                                                                                                                                                           'idProduct': 2616,
                                                                                                                                                                                                                           'idVendor': 1133},
                                                                                                                                                                                                                         { 'IORegistryEntryChildren': [ { 'IORegistryEntryName': 'IOAudioControlUserClient'}],
                                                                                                                                                                                                                           'IORegistryEntryName': 'IOAudioLevelControl'},
[... 358 lines cut ...]
                                                 'DiskUUID': 'AA89E50D-93D9-4106-BC02-6CC3FE921D09',
                                                 'MountPoint': '/Volumes/release',
                                                 'Size': 7086080,
                                                 'VolumeName': 'release',
                                                 'VolumeUUID': 'DC9DC50B-59EA-3E4B-91B7-D7016CC1B168'}],
                               'Size': 7123456}],
  'VolumesFromDisks': [ 'Macintosh HD',
                        'VM',
                        'NODE_F401RE',
                        'MAINTENANCE',
                        'release'],
  'WholeDisks': ['disk0', 'disk1', 'disk2', 'disk3', 'disk4']}
[... cut ...]
+---------------+----------------------+----------------------+------------------------+--------------------------------------------------+-----------------+
| NUCLEO_F401RE | NUCLEO_F401RE[0]     | /Volumes/NODE_F401RE | /dev/tty.usbmodem14433 | 07200221066F6505392BF318                         | 0221            |
| unknown       | None[0]              | /Volumes/MAINTENANCE | /dev/tty.usbmodem14433 | 0000000034544e45003600028aa900102011000097969900 | unknown         |
+---------------+----------------------+----------------------+------------------------+--------------------------------------------------+-----------------+
```

bridadan, 2018-07-13:
Hmm thanks for raising the issue. Looks like another Mac issue. We'll add it to the list.

bridadan, 2019-01-07:
@jupe do you know if this is still an issue?

jupe, 2019-01-10:
Haven’t seen this recently. How much there is changes after this release?

bridadan, 2019-01-10:
> Haven’t seen this recently. How much there is changes after this release?

You can always check the git history 😄  I'm going to close this until we have a case to reproduce this. If you find a way to reproduce please do let us know! We'll happily fix it!

closed 2019-01-10 by bridadan