ARMmbed/mbed-ls #376: device_type is null with version 1.5.0

juhhov, 2018-07-30:
```
$ rm /home/systest/.local/share/mbedls/platforms.json 
$ mbedls --version
WARNING:mbedls.platform_database:Error loading database /home/systest/.local/share/mbedls/platforms.json: Platform Database is out of date; Recreating
1.5.0
$ mbedls -j
[
    {
        "daplink_auto_reset": "0", 
        "daplink_automation_allowed": "1", 
        "daplink_bootloader_crc": "0xb92403e6", 
        "daplink_bootloader_version": "0244", 
        "daplink_daplink_mode": "Interface", 
        "daplink_git_sha": "0beabef8aa4b382809d79e98321ecf6a28936812", 
        "daplink_hic_id": "97969900", 
        "daplink_interface_crc": "0x434eddd1", 
        "daplink_interface_version": "0246", 
        "daplink_local_mods": "0", 
        "daplink_overflow_detection": "0", 
        "daplink_remount_count": "2", 
        "daplink_unique_id": "0240000034544e45004b00018aa9001d2011000097969900", 
        "daplink_usb_interfaces": "MSD, CDC, HID", 
        "daplink_version": "0246", 
        "device_type": null, 
        "directory_entries": [
            "MBED.HTM", 
            "DETAILS.TXT"
        ], 
        "mount_point": "/mnt/pci-0000_00_1d_0-usb-0_1_6_7_1_1_0-scsi-0_0_0_0", 
        "platform_name": "K64F", 
        "platform_name_unique": "K64F[0]", 
        "product_id": "0024", 
        "serial_port": "/dev/ttyACM0", 
        "target_id": "0240000034544e45004b00018aa9001d2011000097969900", 
        "target_id_mbed_htm": "0240000034544e45004b00018aa9001d2011000097969900", 
        "target_id_usb_id": "0240000034544e45004b00018aa9001d2011000097969900", 
        "vendor_id": "8087"
    }, 
    {
        "daplink_build": "Nov 17 2016 13:37:06", 
        "daplink_url": "http://mbed.org/device/?code=079602210A4168153505FEBE", 
        "daplink_version": "0221", 
        "device_type": null, 
        "directory_entries": [
            "DETAILS.TXT", 
            "MBED.HTM"
        ], 
        "mount_point": "/mnt/pci-0000_00_1d_0-usb-0_1_6_7_2_1_1-scsi-0_0_0_0", 
        "platform_name": "NUCLEO_F429ZI", 
        "platform_name_unique": "NUCLEO_F429ZI[0]", 
        "product_id": "0024", 
        "serial_port": "/dev/ttyACM1", 
        "target_id": "079602210A4168153505FEBE", 
        "target_id_mbed_htm": "079602210A4168153505FEBE", 
        "target_id_usb_id": "0675FF494851877267063944", 
        "vendor_id": "8087"
    }
]
```

With v1.4.6 device_type is correct.

bridadan, 2018-07-30:
Oh jeez, of course the update couldn't just work 😟 

Could you run `mbedls --debug` please and post the output?

juhhov, 2018-07-31:
Indeed boards are connected through usb hub and also ykush power switch.

Here is the log:
```
$ mbedls --debug
DEBUG:mbedls.main:mbed-ls ver. 1.5.0
DEBUG:mbedls.main:host: ('posix', 'Linux', '4.4.0-130-generic', '#156-Ubuntu SMP Thu Jun 14 08:53:28 UTC 2018', 'linux2')
DEBUG:mbedls.lstools_linux:Converting device list ['/dev/disk/by-id/usb-MBED_VFS_0240000034544e45004b00018aa9001d2011000097969900-0:0', '/dev/disk/by-id/usb-MBED_microcontroller_0675FF494851877267063944-0:0', '/dev/disk/by-id/usb-Generic-_SM_xD_Picture_058F63626420-0:2', '/dev/disk/by-id/usb-Generic-_MS_MS-Pro_058F63626420-0:3', '/dev/disk/by-id/usb-Generic-_Compact_Flash_058F63626420-0:1', '/dev/disk/by-id/usb-Generic-_SD_MMC_058F63626420-0:0', '/dev/disk/by-id/wwn-0x5001480000000000', '/dev/disk/by-id/ata-HL-DT-ST_DVD+-RW_GT60N_KZYC5MM3148', '/dev/disk/by-id/wwn-0x5002538d412274bd-part1', '/dev/disk/by-id/ata-Samsung_SSD_850_EVO_250GB_S2R6NX0H808412F-part1', '/dev/disk/by-id/wwn-0x5002538d412274bd-part5', '/dev/disk/by-id/ata-Samsung_SSD_850_EVO_250GB_S2R6NX0H808412F-part5', '/dev/disk/by-id/wwn-0x5002538d412274bd-part2', '/dev/disk/by-id/ata-Samsung_SSD_850_EVO_250GB_S2R6NX0H808412F-part2', '/dev/disk/by-id/wwn-0x5002538d412274bd', '/dev/disk/by-id/ata-Samsung_SSD_850_EVO_250GB_S2R6NX0H808412F']
DEBUG:mbedls.lstools_linux:Found disk devices by id {'0240000034544e45004b00018aa9001d2011000097969900': '/dev/sdf', '058F63626420': '/dev/sdb', '0675FF494851877267063944': '/dev/sdg'}
DEBUG:mbedls.lstools_linux:Converting device list ['/dev/serial/by-id/usb-ARM_DAPLink_CMSIS-DAP_0240000034544e45004b00018aa9001d2011000097969900-if01', '/dev/serial/by-id/usb-STMicroelectronics_STM32_STLink_0675FF494851877267063944-if02']
DEBUG:mbedls.lstools_linux:Found serial devices by id {'0240000034544e45004b00018aa9001d2011000097969900': '/dev/ttyACM1', '0675FF494851877267063944': '/dev/ttyACM0'}
DEBUG:mbedls.lstools_linux:Mount mapping {u'/dev/sdf': u'/mnt/pci-0000_00_1d_0-usb-0_1_6_7_1_1_0-scsi-0_0_0_0', u'/dev/sdg': u'/mnt/pci-0000_00_1d_0-usb-0_1_6_7_2_1_1-scsi-0_0_0_0'}
DEBUG:mbedls.lstools_base:Candidates for display [{'vendor_id': '8087', 'mount_point': u'/mnt/pci-0000_00_1d_0-usb-0_1_6_7_1_1_0-scsi-0_0_0_0', 'serial_port': '/dev/ttyACM1', 'target_id_usb_id': '0240000034544e45004b00018aa9001d2011000097969900', 'product_id': '0024'}, {'vendor_id': '8087', 'mount_point': None, 'serial_port': None, 'target_id_usb_id': '058F63626420', 'product_id': '0024'}, {'vendor_id': '8087', 'mount_point': u'/mnt/pci-0000_00_1d_0-usb-0_1_6_7_2_1_1-scsi-0_0_0_0', 'serial_port': '/dev/ttyACM0', 'target_id_usb_id': '0675FF494851877267063944', 'product_id': '0024'}]
DEBUG:mbedls.lstools_base:Found target id 0240000034544e45004b00018aa9001d2011000097969900 in htm line window.location.replace("https://mbed.org/device/?code=0240000034544e45004b00018aa9001d2011000097969900?version=0246?target_id=00000000000000000000000000000000");

DEBUG:mbedls.lstools_base:Found htm target id, 0240000034544e45004b00018aa9001d2011000097969900, for usb target id 0240000034544e45004b00018aa9001d2011000097969900
DEBUG:mbedls.lstools_base:Found target id 079602210A4168153505FEBE in htm line <meta http-equiv="refresh" content="0; url=http://mbed.org/device/?code=079602210A4168153505FEBE"/>

DEBUG:mbedls.lstools_base:Found htm target id, 079602210A4168153505FEBE, for usb target id 0675FF494851877267063944
+---------------+----------------------+------------------------------------------------------+--------------+--------------------------------------------------+-----------------+
| platform_name | platform_name_unique | mount_point                                          | serial_port  | target_id                                        | daplink_version |
+---------------+----------------------+------------------------------------------------------+--------------+--------------------------------------------------+-----------------+
| K64F          | K64F[0]              | /mnt/pci-0000_00_1d_0-usb-0_1_6_7_1_1_0-scsi-0_0_0_0 | /dev/ttyACM1 | 0240000034544e45004b00018aa9001d2011000097969900 | 0246            |
| NUCLEO_F429ZI | NUCLEO_F429ZI[0]     | /mnt/pci-0000_00_1d_0-usb-0_1_6_7_2_1_1-scsi-0_0_0_0 | /dev/ttyACM0 | 079602210A4168153505FEBE                         | 0221            |
+---------------+----------------------+------------------------------------------------------+--------------+--------------------------------------------------+-----------------+
```

closed 2018-07-31 by theotherjimmy

bridadan, 2018-07-31:
@juhhov do you have a minute to try the master branch of mbed-ls on your machine before we make a release? Is that easily doable? I'm pretty confident I have it fixed now but it'd be good to be sure it works on yours as well.

This was unfortunately one of those cases of "works on my machine", so having more data on the different linux setups really helps us make this work better on all machines.

juhhov, 2018-08-01:
Seems to work as expected now! Thanks for a quick fix @bridadan , hopefully we get a patch release with the fix soon :)
```
$ mbedls -j
[
    {
        "daplink_auto_reset": "0", 
        "daplink_automation_allowed": "1", 
        "daplink_bootloader_crc": "0xb92403e6", 
        "daplink_bootloader_version": "0244", 
        "daplink_daplink_mode": "Interface", 
        "daplink_git_sha": "0beabef8aa4b382809d79e98321ecf6a28936812", 
        "daplink_hic_id": "97969900", 
        "daplink_interface_crc": "0x434eddd1", 
        "daplink_interface_version": "0246", 
        "daplink_local_mods": "0", 
        "daplink_overflow_detection": "0", 
        "daplink_remount_count": "1", 
        "daplink_unique_id": "0240000034544e45004b00018aa9001d2011000097969900", 
        "daplink_usb_interfaces": "MSD, CDC, HID", 
        "daplink_version": "0246", 
        "device_type": "daplink", 
        "directory_entries": [
            "MBED.HTM", 
            "DETAILS.TXT"
        ], 
        "mount_point": "/mnt/pci-0000_00_1d_0-usb-0_1_6_7_1_1_0-scsi-0_0_0_0", 
        "platform_name": "K64F", 
        "platform_name_unique": "K64F[0]", 
        "product_id": "0204", 
        "serial_port": "/dev/ttyACM1", 
        "target_id": "0240000034544e45004b00018aa9001d2011000097969900", 
        "target_id_mbed_htm": "0240000034544e45004b00018aa9001d2011000097969900", 
        "target_id_usb_id": "0240000034544e45004b00018aa9001d2011000097969900", 
        "vendor_id": "0d28"
    }, 
    {
        "daplink_build": "Nov 17 2016 13:37:06", 
        "daplink_url": "http://mbed.org/device/?code=079602210A4168153505FEBE", 
        "daplink_version": "0221", 
        "device_type": "stlink", 
        "directory_entries": [
            "DETAILS.TXT", 
            "MBED.HTM"
        ], 
        "mount_point": "/mnt/pci-0000_00_1d_0-usb-0_1_6_7_2_1_1-scsi-0_0_0_0", 
        "platform_name": "NUCLEO_F429ZI", 
        "platform_name_unique": "NUCLEO_F429ZI[0]", 
        "product_id": "374b", 
        "serial_port": "/dev/ttyACM0", 
        "target_id": "079602210A4168153505FEBE", 
        "target_id_mbed_htm": "079602210A4168153505FEBE", 
        "target_id_usb_id": "0675FF494851877267063944", 
        "vendor_id": "0483"
    }
]
```

bridadan, 2018-08-01:
Awesome, thanks for testing!

PR #377 (2018-07-31): Linux vid pid detection fix
This should fix #376 as well as an issue discovered by the DAPLink CI.

This accounts for the case where mbeds are connected to the system through (potentially multiple) USB hubs.

For more information on the USB sysfs paths, view the section "USB and Sysfs" on this page: https://www.safaribooksonline.com/library/view/linux-device-drivers/0596005903/ch13.html.

FYI @flit @juhhov