ARMmbed/mbed-ls #413: It is not good to list all jlink opensda device

Hoohaha, 2019-01-07:
Hi,

As far as i I know mebdls has supported the jlink detection.
But it can not list  jlink OpenSDA device for NXP kinetis on windows.

If I open JLink.exe, and input "ShowEmuList" command, i can see it.

[image]

bridadan, 2019-01-07:
Hi @Hoohaha, mbed-ls does support detecting jlink devices, however each target needs to be enabled individually so we can detect it. You can see the currently enabled jlink devices here: https://github.com/ARMmbed/mbed-ls/blob/master/mbed_lstools/platform_database.py#L320-L341

Which target are you trying to detect?

Hoohaha, 2019-01-08:
At the begining, i wronly thought that mbedls is a tool to help to list embedded devices, but as you mentioned this just for mbed os. So it only working for some specific boards, is right?

If yes, i think i can close #412 and this one.

bridadan, 2019-01-08:
> So it only working for some specific boards, is right?

That's correct. If you'd like to add support for your target for your own purposes you can add it temporarily using mocking: https://github.com/ARMmbed/mbed-ls#mocking-renaming-platforms

Or you can add it in your own fork by following these instructions: https://github.com/ARMmbed/mbed-ls#adding-platform-support

closed 2019-01-08 by bridadan