ARMmbed/mbed-ls #414: adding non-development-board target

geoffrey-vl, 2019-01-07:
We have a custom device that's not mainline in mbed-os. It's derived from a NUCLEO_F429ZI, so we tried having it detect as one by setting the detect_code in target.json to 0796.

When we use the ST-LINK/V2 programmer our board is not detected. However if we take a NUCLEO_F429ZI, knock-off its ST-LINK/V2 programmer and hook that programmer to our board mbed-ls seems to work perfectly fine.

Can we make it work with the standalone programmer too so that greentea can kick in?

bridadan, 2019-01-07:
Hi @geoffrey-vl, does the st-link/v2 programmer have a mass storage device (drag and drop programming) and a serial port? If so it may be possible! This is the reason why using a modified NUCLEO_F429ZI board works with mbed-ls.

geoffrey-vl, 2019-01-07:
hi @bridadan 
No it hasn't, it seems to have different firmware.
Can't we make the flashing part work through pyocd for instance?
And what about the ability to make "generic" device supported for mbed-ls where we can specify the COM-PORT ourselves for either flashing and/or debug?

bridadan, 2019-01-07:
Hi @geoffrey-vl, thanks for the info, that's what I thought but I wanted to be sure.

> Can't we make the flashing part work through pyocd for instance?

The test tools do have initial support for flashing through pyocd, however they still rely on mbed-ls to detect the target in the first place. This isn't ideal and I'd like to revisit this in the future. However, it would require quite a bit of rework within mbed-ls/detect to get this working smoothly.

> And what about the ability to make "generic" device supported for mbed-ls where we can specify the COM-PORT ourselves for either flashing and/or debug?

You can specify the COM port for serial communication and mount point for flashing with retargeting: https://github.com/ARMmbed/mbed-ls/#retargeting-platforms

But this doesn't solve the case when the target doesn't have a mass storage device. This is unfortunately just a gap in our tools for the moment. Though again, I'd certainly like to revisit this issue in the future!

geoffrey-vl, 2019-01-07:
Indeed, solving that would enable any non development board to use greentea, I'd be glad to pick up on that

closed 2019-01-07 by geoffrey-vl