ARMmbed/mbed-ls #416: List ST boards under "stlink" section in DEFAULT_PLATFORM_DB

RomanSaveljev, 2019-01-08:
I was wondering why ST-boards are listed under "daplink" section in the `DEFAULT_PLATFORM_DB`? The `device_type` seems to be correctly resolving to "stlink" from VID. Turns out mbed-flasher depends on the right content in `DEFAULT_PLATFORM_DB`, when choosing the right flashing method.

bridadan, 2019-01-08:
Hi @RomanSaveljev, those boards are under the "daplink" section for legacy reasons I suppose. The detection "method" for both daplink and stlink are the same (read contents from mbed.htm and details.txt).

Typically we'd recommend you to run the `list_mbeds()` api and use that information instead of pulling the big list of platforms. Can you use the information provided by this API, or is this information not available in mbed-flasher?

RomanSaveljev, 2019-01-08:
I was hoping to get away with a simple patch (https://github.com/ARMmbed/mbed-flasher/pull/153). However, from what you are saying it seems like a bigger refactoring is in order...